Tibus/TouchScreen-For-NanoDLP #3: Attempting to use with Pi3 - Nextion hangs on "connecting to printer...."

sgraber, 2017-09-05:
I've followed all of the steps to have the display work with the Nextion 2.4" display, but the display hangs on "connecting to printer..."  Is there any sort of log I can be looking at to see why it's hanging?  Also:  I'm assuming that I am to be enabling the login shell over serial from raspi-config (Advanced -> Serial -> Yes)?  Or should this be set to No?

diginfo, 2017-10-12:
Hi, I have the same problem, all steps followed (not difficult !) and all I get is connecting to printer.

Did you solve this ?

Tibus, 2017-10-12:
did you try this ? 
https://www.nanodlp.com/forum/viewtopic.php?id=710

diginfo, 2017-10-12:
Thanks, yes I did, but it did not work for me, b ut I am using original RPI3 (not B).

Here's how I got it to work:

## get the pi3 overlays
wget https://github.com/PiNet/PiNet-Boot/blob/master/boot/overlays/pi3-disable-bt-overlay.dtb
mv pi3-disable-bt-overlay.dtb /boot/overlays

wget https://github.com/PiNet/PiNet-Boot/blob/master/boot/overlays/pi3-miniuart-bt-overlay.dtb
mv pi3-miniuart-bt-overlay.dtb /boot/overlays

# edit /boot/config.txt & append:
dtoverlay=pi3-disable-bt
dtoverlay=pi3-miniuart-bt-overlay
force_turbo=1

Tibus, 2017-10-12:
Is it working now by doing this?
I have to write a bash file to configure in one step all you need on raspberry pi 3/pizero/pi3B,...

diginfo, 2017-10-12:
Yes, it's working now, but I also did and un-did a lot of things over the last few hours during my debug process, however these steps did make it work.

Before that I got the following errors in the output.log:

`list [ { comName: '/dev/ttyAMA0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: '/dev/ttyS0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined } ]
error opening port  /dev/ttyAMA0 retry in 2 seconds`

There might be a better source for the overlay files, but they worked.

diginfo, 2017-10-12:
Oh yes, also when loading the .tft file, I got an error something like "wrong model" - this is because I am using the advanced version with the built in RTC.

I re-compiled changing the model number to the advanced and used this file to load.

I saw another user on the forum reporting the same error, might be worthwhile adding comments or notes for this.

In your original readme, you stated 2.4" but you did not indicate standard or advanced version, I did not know that it made a difference when I ordered the screen !

Tibus, 2017-10-12:
Yes, I didn't have advanced version and tought there was no error running the standard .tft file on an advanced screen. I'll add .tft for every version in the future

diginfo, 2017-10-12:
It would also be nice (if you have the time) to design a version that emulates the new wanhao box :-)

Tibus, 2017-10-12:
For the UX part?
Yes, I'm planning to redo all the UX. Do you have a video of the wanhao box screen UX?

jamesarm97, 2017-10-12:
Emulates or "looks" like the wanhao UI, because if you emulate it, it will be a step backwards ;)

jamesarm97, 2017-10-12:
Manual http://www.wanhao3dprinter.com/xiazai/OQD08011_D7_Box.pdf

jamesarm97, 2017-10-12:
The main menu / icons look good but everything past that is very basic and plane. No image preview / layer preview or anything.

diginfo, 2017-10-12:
The level of control is great, but the UI design can be a bit 'fiddly' if you have big sausage fingers or your eyesight is not so good ;-)

Tibus, 2017-10-12:
yes, I 'll never emulate it ;D just take best points and try to reproduce it. I see a video, it's nice but there is no image/print preview,... but it's nice, I'll do my best to make a better UX for nanoDLP/nextion ;D

Dylan144GT, 2017-10-12:
Hi Tibus,

Have you managed to interface to USB yet? I am busy doing a similar thing to you except I am using Python instead of NodeJS. The reason I ask is because I am having trouble submitting a form to nanoDLP with a post request. I get a 200 response when I post but nothin happens. Any ideas?

Thanks,

Dylan

Tibus, 2017-10-12:
Hi,
Not done yet, 'll work on it in few weeks.

Dylan144GT, 2017-10-12:
Okay, thanks. Thanks for a good project BTW! :)

Maxjewels, 2017-10-12:
I use the advanced version 3.2 "and I did not have any problem

diginfo, 2017-10-18:
Had to do a re-install from scratch using release 1437 as I am using the wanhao D7 and are unable to get past the "connecting to Printer' again.

I have checked and repeated all the steps that are listed in the forum but still nogo, not sure if it is a compatability issue with V1437 ?

There's nothing obvious in the output.log file (previously it was unable to connect)

`{ nextion: { port: '/dev/ttyAMA0' },
  pushbullet: { apiKey: null } } { apiKey: null }
init
connect to port /dev/ttyAMA0
list [ { comName: '/dev/ttyACM0',
    manufacturer: 'Arduino__www.arduino.cc_',
    serialNumber: 'Arduino__www.arduino.cc__Arduino_Mega_2560_7563331323335191C181',
    pnpId: 'usb-Arduino__www.arduino.cc__Arduino_Mega_2560_7563331323335191C181-if00',
    vendorId: '0x2341',
    productId: '0x0010' },
  { comName: '/dev/ttyAMA0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: '/dev/ttyS0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined } ]
`

I have tried running the node app directly from the console, but I get no errors, only the output as above, how can I troubleshoot this ?

Dylan144GT, 2017-11-07:
Hi Tibus,

Can you please tell me how you formulate the time remaining of a print? Ie. what information you fetch from where and the formula you apply with that information in order to calculate the remaining time of a print.

Thanks for the help.

Dylan

Tibus, 2017-11-07:
Hello,
The time remaining is calculated by NanoDLP, not my software.

Dylan144GT, 2017-11-07:
Thanks for the response, I found what I was looking for though. It was in "PrintingHome.js" line 51 and 52.
Thanks!

Tibus, 2017-11-07:
Oups, sorry, I didn't remember Nanodlp do not send time remaining in minute or so. So I take the formula from NanoDLP Javascript home page.

sgraber, 2017-11-07:
@Tibus : Is there a way to get debugging information when trying to connect your Nextion software to the Pi?  I added four buttons to the Home screen, adjusted the js accordingly (I think?), compiled it, installed it.  I'm now trying to debug why my custom buttons are not working.  I'd like to see what calls are being made when I push a button.  Is there a Pi debugging console somewhere?

Tibus, 2017-11-07:
I never debug on the raspberry.
If you run `node bin/index.js` on your computer, with the ip of your nanodlp computer in the config file ans the nextion screen connected to your computer (with so the good COM port in the config), the screen will manage NanoDLP trough the ethernet and you will be able to debug it.

Elsewhere, you can pkill node and run `DEBUG=* node bin/index.js` inside the nextion folder on your raspberry pi to get the debugging console on your raspberry

jcrorxp, 2017-12-13:
@sgraber  i think you need to add some Page file ？

BeardyWargameSi, 2018-01-20:
HI I have a Nextion 3.5" Screen - NX4832T035 and this just is not working at all. Tried the steps I don't know how many times in the last 24 hours, including the long walkthrough on the Tips and Tricks nanoDLP page, nothing. :/

I've updated to the latest stable build of nanoDLP, etc etc.

Honestly at my wits end with this and any help would be greatly appreciated.

The screen is stuck on a connecting to printer loop

Thanks

setal2003, 2018-01-22:
Hello there 
I have done all the steps and it still doesn't work
at the end I get the message of can not unlink certain directories
Please help!
thank you
> pi@raspberrypi:~ $ mkdir /home/pi/nextion;(wget https://github.com/Tibus/TouchSc           reen-For-NanoDLP/releases/download/V0.2/pi2_3.tar.gz --no-check-certificate -O -            | tar -C /home/pi/nextion -xz);cd /home/pi/nextion; sudo ./setup.sh
> --2018-01-22 22:37:22--  https://github.com/Tibus/TouchScreen-For-NanoDLP/releas           es/download/V0.2/pi2_3.tar.gz
> Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
> Connecting to github.com (github.com)|192.30.253.113|:443... connected.
> HTTP request sent, awaiting response... 302 Found
> Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/963614           42/9c96c61a-be85-11e7-807d-eba7033ee479?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-C           redential=AKIAIWNJYAX4CSVEH53A%2F20180122%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-           Date=20180122T223723Z&X-Amz-Expires=300&X-Amz-Signature=d051cdb3832d626dff2ac94f           97d5b6c844f5a7f8a5c7ae5e5a85e5d9cdb45025&X-Amz-SignedHeaders=host&actor_id=0&res           ponse-content-disposition=attachment%3B%20filename%3Dpi2_3.tar.gz&response-conte           nt-type=application%2Foctet-stream [following]
> --2018-01-22 22:37:23--  https://github-production-release-asset-2e65be.s3.amazo           naws.com/96361442/9c96c61a-be85-11e7-807d-eba7033ee479?X-Amz-Algorithm=AWS4-HMAC           -SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180122%2Fus-east-1%2Fs3%2Faws4           _request&X-Amz-Date=20180122T223723Z&X-Amz-Expires=300&X-Amz-Signature=d051cdb38           32d626dff2ac94f97d5b6c844f5a7f8a5c7ae5e5a85e5d9cdb45025&X-Amz-SignedHeaders=host           &actor_id=0&response-content-disposition=attachment%3B%20filename%3Dpi2_3.tar.gz           &response-content-type=application%2Foctet-stream
> Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-produc           tion-release-asset-2e65be.s3.amazonaws.com)... 54.231.81.242
> Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-pr           oduction-release-asset-2e65be.s3.amazonaws.com)|54.231.81.242|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 26420532 (25M) [application/octet-stream]
> Saving to: ‘STDOUT’
>
> -                      100%[=============================>]  25.20M  1003KB/s   in 22s
>
> 2018-01-22 22:37:46 (1.14 MB/s) - written to stdout [26420532/26420532]
>
> --2018-01-22 22:37:46--  https://raw.githubusercontent.com/sdesalas/node-pi-zero/master/install-node-v6.4.0.sh
> Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.112.133
> Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.112.133|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 1324 (1.3K) [text/plain]
> Saving to: ‘STDOUT’
>
> -                      100%[=============================>]   1.29K  --.-KB/s   in 0s
>
> 2018-01-22 22:37:46 (15.7 MB/s) - written to stdout [1324/1324]
>
> --2018-01-22 22:37:46--  https://nodejs.org/dist/v6.4.0/node-v6.4.0-linux-armv6l.tar.gz
> Resolving nodejs.org (nodejs.org)... 2400:cb00:2048:1::6814:172e, 2400:cb00:2048:1::6814:162e, 104.20.23.46, ...
> Connecting to nodejs.org (nodejs.org)|2400:cb00:2048:1::6814:172e|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 12746855 (12M) [application/gzip]
> Saving to: ‘node-v6.4.0-linux-armv6l.tar.gz’
>
> node-v6.4.0-linux-armv 100%[=============================>]  12.16M  1.79MB/s   in 6.8s   s
>
> 2018-01-22 22:37:54 (1.79 MB/s) - ‘node-v6.4.0-linux-armv6l.tar.gz’ saved [12746855/12746855]
>
> unlink: cannot unlink ‘/usr/bin/node’: No such file or directory
> unlink: cannot unlink ‘/usr/sbin/node’: No such file or directory
> unlink: cannot unlink ‘/sbin/node’: No such file or directory
> unlink: cannot unlink ‘/usr/local/bin/node’: No such file or directory
> unlink: cannot unlink ‘/usr/bin/npm’: No such file or directory
> unlink: cannot unlink ‘/usr/sbin/npm’: No such file or directory
> unlink: cannot unlink ‘/sbin/npm’: No such file or directory
> unlink: cannot unlink ‘/usr/local/bin/npm’: No such file or directory
> pi@raspberrypi:~/nextion $
>

setal2003, 2018-01-22:
All I get is connecting to printer message

BeardyWargameSi, 2018-01-23:
I got this working over the weekend, by using this https://github.com/cuboid3d/nanodlp just follow the instructions - I downloaded the .rar and unzipped it over cloning the git

jamesarm97, 2018-01-23:
Did you just use the script part because the actual fork of Tibus may be old.

rdastartupguy, 2018-05-23:
Still stuck at connecting to printer message on Rpi 2 model B+ connected to nextion 3.5 :( 

ErrorOutput.log shows  
"missing API key for pushbullet" 

No other errors :S

hu5566798, 2018-06-14:
@Tibus Is it ok for the Pi3B, I use PI3B follow the step.  it will appear the message as setal2003
unlink: cannot unlink ‘/usr/bin/node’: No such file or directory
unlink: cannot unlink ‘/usr/sbin/node’: No such file or directory
unlink: cannot unlink ‘/sbin/node’: No such file or directory
unlink: cannot unlink ‘/usr/local/bin/node’: No such file or directory
unlink: cannot unlink ‘/usr/bin/npm’: No such file or directory
unlink: cannot unlink ‘/usr/sbin/npm’: No such file or directory
unlink: cannot unlink ‘/sbin/npm’: No such file or directory
unlink: cannot unlink ‘/usr/local/bin/npm’: No such file or directory

but after redo 
pi@raspberrypi:~ $ mkdir /home/pi/nextion;(wget https://github.com/Tibus/TouchSc reen-For-NanoDLP/releases/download/V0.2/pi2_3.tar.gz --no-check-certificate -O - | tar -C /home/pi/nextion -xz);cd /home/pi/nextion; sudo ./setup.sh
reboot PI3B  the nextion can be show the botton and image. but each botton can't response with touch. whatever you touch the botton will not response.  what's wrong？

hu5566798, 2018-06-14:
I also update the v0.3 tft to my 3.2'' nextion but can't show the IP/TCP code.

jmtru, 2018-06-19:
@Tibus  , I'm in the same situation as @hu5566798 .

can you help us? 

my screen is the model nx8048k070

Thank you

hu5566798, 2018-06-20:
@ jmtru， found out the solution below:
first see the errorlog(sudo cd /home/pi/nextion, sudo ls -al, you will see that errorlog in list ) in nextion， if it is below error:
missing API key for pushbullet
{ Error: Cannot find module 'ip'
    at Function.Module._resolveFilename (module.js:455:15)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at Home._callee$ (/home/pi/nextion/bin/plugins/nextion/pages/home.js:54:22)
    at tryCatch (/home/pi/nextion/node_modules/regenerator-runtime/runtime.js:65:40)
    at GeneratorFunctionPrototype.invoke [as _invoke] (/home/pi/nextion/node_modules/regenera$
    at GeneratorFunctionPrototype.prototype.(anonymous function) [as next] (/home/pi/nextion/$
    at step (/home/pi/nextion/bin/plugins/nextion/pages/home.js:16:191)
    at /home/pi/nextion/bin/plugins/nextion/pages/home.js:16:361 code: 'MODULE_NOT_FOUND' }

The new dependencies (ip, drivelist, child_process) are missing in release v 0.3
sudo cd /home/pi/nextion
sudo npm i ip
sudo npm i drivelist
sudo npm i child_process
sudo shutdown -r now
after reboot the IP/TCP WILL have address, all the set icon can be work. 
Thank for ToS reply on the NanoDLP forum below link:
https://www.nanodlp.com/forum/viewtopic.php?id=710&p=6

hu5566798, 2018-06-20:
@ jmtru after reboot, you will find the IP address is strange like 127.0.0.1,  because it loss it wifi.  you can reconnect with the net cable. after connect net cable you can find it change into http://169.254.187.XX, just fill this address into your web Explorer, then goto the home page at the right top  you will find the wifi logo, juts click into choose your wifi with password.  then reboot and remove your net cable, then your wifi connect IP address will come back.

hu5566798, 2018-06-20:
@jmtru, It show that TOS and Tibus is working V0.4 to fix that promble. but I also want to know why itead/Segnix can't work on PI3B+nextion? anyone try that?
https://github.com/itead/Segnix

Tibus, 2018-06-20:
127.0.0.1 is the localhost ip so no need to change it!

hu5566798, 2018-06-20:
@Tibus Ok, thanks. hope you can release new version to fix that problem. (ip, drivelist, child_process) are missing in release v 0.3

jmtru, 2018-06-20:
**pi@raspberrypi:~ $** sudo cd /home/pi/nextion
sudo: cd: command not found
**pi@raspberrypi:~ $**

jmtru, 2018-06-20:
Sorry. My english is very bad....

pi@raspberrypi:~ $ cd /home/pi/nextion
pi@raspberrypi:~/nextion $ npm i ip
/home/pi/nextion
└── ip@1.1.5

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/nextion/package.json'
npm WARN nextion No description
npm WARN nextion No repository field.
npm WARN nextion No README data
npm WARN nextion No license field.
pi@raspberrypi:~/nextion $ npm i drivelist

> drivelist@6.2.2 install /home/pi/nextion/node_modules/drivelist
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=6.4.0 runtime=node arch=arm platform=linux)
make: Entering directory '/home/pi/nextion/node_modules/drivelist/build'
  CXX(target) Release/obj.target/drivelist/src/drivelist.o
../src/drivelist.cpp: In member function ‘virtual void DriveListWorker::HandleOKCallback()’:
../src/drivelist.cpp:44:27: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
     callback->Call(2, argv);
                           ^
In file included from ../src/drivelist.cpp:17:0:
../node_modules/nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
  CXX(target) Release/obj.target/drivelist/src/device-descriptor.o
  CXX(target) Release/obj.target/drivelist/src/linux/list.o
  SOLINK_MODULE(target) Release/obj.target/drivelist.node
  COPY Release/drivelist.node
make: Leaving directory '/home/pi/nextion/node_modules/drivelist/build'
/home/pi/nextion
└─┬ drivelist@6.2.2
  ├── bindings@1.3.0
  ├── debug@3.1.0
  ├── fast-plist@0.1.2
  ├─┬ js-yaml@3.12.0
  │ ├─┬ argparse@1.0.10
  │ │ └── sprintf-js@1.0.3
  │ └── esprima@4.0.0
  ├── nan@2.10.0
  └─┬ prebuild-install@4.0.0
    ├── detect-libc@1.0.3
    ├── expand-template@1.1.1
    ├── github-from-package@0.0.0
    ├─┬ node-abi@2.4.1
    │ └── semver@5.5.0
    ├── noop-logger@0.1.1
    ├─┬ npmlog@4.1.2
    │ ├─┬ are-we-there-yet@1.1.5
    │ │ ├── delegates@1.0.0
    │ │ └─┬ readable-stream@2.3.6
    │ │   ├── core-util-is@1.0.2
    │ │   ├── inherits@2.0.3
    │ │   ├── isarray@1.0.0
    │ │   ├── process-nextick-args@2.0.0
    │ │   ├── string_decoder@1.1.1
    │ │   └── util-deprecate@1.0.2
    │ ├── console-control-strings@1.1.0
    │ ├─┬ gauge@2.7.4
    │ │ ├── aproba@1.2.0
    │ │ ├── has-unicode@2.0.1
    │ │ ├── object-assign@4.1.1
    │ │ ├── signal-exit@3.0.2
    │ │ ├─┬ string-width@1.0.2
    │ │ │ ├── code-point-at@1.1.0
    │ │ │ └─┬ is-fullwidth-code-point@1.0.0
    │ │ │   └── number-is-nan@1.0.1
    │ │ └── wide-align@1.1.3
    │ └── set-blocking@2.0.0
    ├── os-homedir@1.0.2
    ├─┬ pump@2.0.1
    │ ├── end-of-stream@1.4.1
    │ └─┬ once@1.4.0
    │   └── wrappy@1.0.2
    ├─┬ simple-get@2.8.1
    │ └── simple-concat@1.0.0
    ├─┬ tar-fs@1.16.3
    │ ├── chownr@1.0.1
    │ ├── pump@1.0.3
    │ └─┬ tar-stream@1.6.1
    │   ├── bl@1.2.2
    │   ├─┬ buffer-alloc@1.2.0
    │   │ ├── buffer-alloc-unsafe@1.1.0
    │   │ └── buffer-fill@1.0.0
    │   ├── fs-constants@1.0.0
    │   ├── to-buffer@1.1.1
    │   └── xtend@4.0.1
    └── which-pm-runs@1.0.0

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/nextion/package.json'
npm WARN nextion No description
npm WARN nextion No repository field.
npm WARN nextion No README data
npm WARN nextion No license field.
pi@raspberrypi:~/nextion $ npm i child_process
/home/pi/nextion
└── child_process@1.0.2

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/nextion/package.json'
npm WARN nextion No description
npm WARN nextion No repository field.
npm WARN nextion No README data
npm WARN nextion No license field.

jmtru, 2018-06-20:
works!!! thanks!!!!

cybertec02, 2018-08-02:
Hi guys, i got the screen to display the menus but it wont respond to any input. Top shows "not printing" then in the middle tcp/ip. Anything i missed? Nextion 3.2 raspi3 B

rdastartupguy, 2018-08-02:
hey Cybertec02
I am having the same problem where it shows the UI but doesn't respond to
input. Not sure how to solve that, but as for your TCP/IP text field, you
need to install a small TCP module. I will share the code when I reach home.

hu5566798, 2018-08-02:
@cybertec02 
try below command
sudo cd /home/pi/nextion
sudo npm i ip
sudo npm i drivelist
sudo npm i child_process
sudo shutdown -r now
after reboot the IP/TCP WILL have address, all the set icon can be work.

cybertec02, 2018-08-04:
@hu5566798 
Thanks, it worked.

probotix, 2019-11-02:
Have tried all of this on a nextion 3.5 enhanced, i finally got the screen to display (no longer stuck on connecting to printer), but the the buttons still don't work. I am using ttyUSB0.

stefi01, 2019-11-25:
same for me, i do get a GUI on the display but pressing any button produces a dark highlight on the screen as i am pressing but no more happens

probotix, 2019-11-25:
I solved it here: https://www.nanodlp.com/forum/viewtopic.php?id=2196

Problem you are seeing is probably because you are using the wrong tft file. The ones on github did not work for me, the ones on the nandlp.com website did.