Tibus/TouchScreen-For-NanoDLP #6: Z-axis screen Home button not working

EberleG, 2018-01-28:
Hi, I have found the home button on the Z-axis control screen/page doesn't work.  All the z-axis controls still function but I can't leave that page.  BTW, I followed the previous issue post (solved: "connecting to printer" issue) to get my Nextion screen (NX4832T035) working on a Pi3 B.  I'm not sure that introduced this bug?

mylove29985, 2018-04-23:
I have the same problem. Home button on the Z axis control not work. Reboot and Shutdow also not working. I am using 2.4 nextion screen

SteBuTOS, 2018-04-23:
Sounds to me like you only updated the tft-file on the nextion but did not update the javascript code (home botton issue on z-axis screen requires fixed zAxis.js in bin folder, also new features require corresponding changes/ additions in javascript files.

mylove29985, 2018-04-23:
Thanks for quick respond. Can you guide me how to upgrade this javascrip? Upgrade via Rasberry terminal or in nextionsreen. Cuz I am a new in this so I dont know how :(

SteBuTOS, 2018-04-23:
Tibus currently preparing an official 3rd release.
Until that becomes available you might try the following:
mkdir /home/pi/nextion;(wget https://github.com/SteBuTOS/TouchScreen-For-NanoDLP/archive/v0.3.tar.gz --no-check-certificate -O - | tar -C /home/pi/nextion -xz);cd /home/pi/nextion; sudo ./setup.sh

mylove29985, 2018-04-23:
Thanks alot. I will try it, waiting to the nex release :D

mylove29985, 2018-04-23:
I tried this as you recomended but no luck, till the same problem???
mkdir /home/pi/nextion;(wget https://github.com/SteBuTOS/TouchScreen-For-NanoDLP/archive/v0.3.tar.gz --no-check-certificate -O - | tar -C /home/pi/nextion -xz);cd /home/pi/nextion; sudo ./setup.sh

SteBuTOS, 2018-04-23:
Did you copy the complete sequence at once to the command line and executed it? If so, did you get any error messages? You might give it a second try with „sudo“ in front

SteBuTOS, 2018-04-23:
Most importand: once installed, do a reboot!

mylove29985, 2018-04-23:
Yes i coppied and executed it. it show some faul mean " folder exist" then it download ( about 12Mb) then ended. After that I do a reboot. I will try again with "sudo" in front and take the picture.

mylove29985, 2018-04-23:
I tried again with no luck. it show error (please see 2 pictures)
https://upanhtocdo.com/image/Sotg3
https://upanhtocdo.com/image/Sonpv

SteBuTOS, 2018-04-23:
(Last) Try:
cd /home/pi/nextion;sudo (wget https://github.com/SteBuTOS/TouchScreen-For-NanoDLP/archive/v0.3.tar.gz --no-check-certificate -O - | tar -C /home/pi/nextion -xz);cd /home/pi/nextion; sudo ./setup.sh

mylove29985, 2018-04-23:
It show error in command line:
bash: syntax error near unexpected token `wget'

SteBuTOS, 2018-04-23:
sudo cd /home/pi/nextion;(wget https://github.com/SteBuTOS/TouchScreen-For-NanoDLP/archive/v0.3.tar.gz --no-check-certificate -O - | tar -C /home/pi/nextion -xz);cd /home/pi/nextion; sudo ./setup.sh

SteBuTOS, 2018-04-23:
Right at the moment I have no access to linux machine for testing myself. So it becomes a trial an error which is not very effectlve. Sorry for that

mylove29985, 2018-04-23:
Hi I tried but the same problems as two pictures I showed. Maybe i have to reinstall all. Thanks alot for support.

SteBuTOS, 2018-04-23:
ok, had some time on my hand on a linux machine. the following script shourd work:

`sudo wget https://github.com/SteBuTOS/TouchScreen-For-NanoDLP/archive/v0.3.tar.gz --no-check-certificate -O - | sudo tar  -C /home/pi/nextion -xz`

`sudo cp /home/pi/nextion/TouchScreen-For-NanoDLP-0.3/. /home/pi/nextion/. -rv`

`sudo rm -r /home/pi/nextion/TouchScreen-For-NanoDLP-0.3`

`sudo npm i ip`

`sudo npm i drivelist`

`sudo npm i child_process`

`sudo shutdown -r now`

mylove29985, 2018-04-24:
Thanks you for help me. I tried and the Shutdown and reboot button is working now!!! But Home button on Z axis calibration till not work and in this section I can choice arrange of number like the picture. Some thing went wrong ???
https://upanhtocdo.com/image/SUfw5

SteBuTOS, 2018-04-24:
The script also downloaded a fresh set of tft files. Can you try to reinstall the right one to your display with sd card?!

mylove29985, 2018-04-24:
I tried but nothing. What the different between 2.4.tft and 2.4 tft enhance. I try to install 2.4 enhance tft file to my screen but it showed wrong file.

SteBuTOS, 2018-04-24:
Can you check for a file errorOutput.log in your nextion folder on your Pi. If the content is not to huge, pist it here.

SteBuTOS, 2018-04-24:
Err: pist=post

mylove29985, 2018-04-25:
Hi, I checked errorOutput.log, only 1 line "missing API key for pushbullet" how to solve that? Thanks

SteBuTOS, 2018-04-25:
That message seems not related to your issue with the zAxis screen. Your screenshot indicates that in the javascript zAxis.js in the bin-subfolder the function setBtn() was not executed. Thats strange, bacuse in that area was no code chang lately, so it‘s more likely that there is something that prevents that file from beeing executed in general in your setup. Can you please check the following file:
/home/pi/nextion/bin/plugins/nextion/pages/zAxis.js
Two questions here:
1 does it exist?
2 whats the content?

mylove29985, 2018-04-25:
Yes it exist, I copy the content as below

"use strict";

Object.defineProperty(exports, "__esModule", {
  value: true
});
exports.default = undefined;

var _createClass = function () { function defineProperties(target, props) { for (var i = 0; i < props.length; i++) { var descriptor = props[i]; descriptor.enumerable = descriptor.enumerable || false; descriptor.configurable = true; if ("value" in descriptor) descriptor.writable = true; Object.defineProperty(target, descriptor.key, descriptor); } } return function (Constructor, protoProps, staticProps) { if (protoProps) defineProperties(Constructor.prototype, protoProps); if (staticProps) defineProperties(Constructor, staticProps); return Constructor; }; }();

var _abstract2 = require("./abstract.js");

var _abstract3 = _interopRequireDefault(_abstract2);

function _interopRequireDefault(obj) { return obj && obj.__esModule ? obj : { default: obj }; }

function _asyncToGenerator(fn) { return function () { var gen = fn.apply(this, arguments); return new Promise(function (resolve, reject) { function step(key, arg) { try { var info = gen[key](arg); var value = info.value; } catch (error) { reject(error); return; } if (info.done) { resolve(value); } else { return Promise.resolve(value).then(function (value) { step("next", value); }, function (err) { step("throw", err); }); } } return step("next"); }); }; }

function _classCallCheck(instance, Constructor) { if (!(instance instanceof Constructor)) { throw new TypeError("Cannot call a class as a function"); } }

function _possibleConstructorReturn(self, call) { if (!self) { throw new ReferenceError("this hasn't been initialised - super() hasn't been called"); } return call && (typeof call === "object" || typeof call === "function") ? call : self; }

function _inherits(subClass, superClass) { if (typeof superClass !== "function" && superClass !== null) { throw new TypeError("Super expression must either be null or a function, not " + typeof superClass); } subClass.prototype = Object.create(superClass && superClass.prototype, { constructor: { value: subClass, enumerable: false, writable: true, configurable: true } }); if (superClass) Object.setPrototypeOf ? Object.setPrototypeOf(subClass, superClass) : subClass.__proto__ = superClass; }

require("babel-polyfill");

var ZAxis = function (_abstract) {
  _inherits(ZAxis, _abstract);

  function ZAxis(screenManager) {
    _classCallCheck(this, ZAxis);

    var _this = _possibleConstructorReturn(this, (ZAxis.__proto__ || Object.getPrototypeOf(ZAxis)).call(this, screenManager));

    _this.currentButton = 8;
    return _this;
  }

  _createClass(ZAxis, [{
    key: "init",
    value: function () {
      var _ref = _asyncToGenerator( /*#__PURE__*/regeneratorRuntime.mark(function _callee(options) {
        var _this2 = this;

        return regeneratorRuntime.wrap(function _callee$(_context) {
          while (1) {
            switch (_context.prev = _context.next) {
              case 0:
                _context.next = 2;
                return this.setScreen("zAxis");

              case 2:

                this.addListener("click_b16", function (e) {
                  _this2.changePage("home");
                });

                this.addListener("click_b12", function (e) {
                  _this2.up();
                });

                this.addListener("click_b13", function (e) {
                  _this2.down();
                });

                this.addListener("click_b11", function () {
                  return _this2.setBtn(11);
                });
                this.addListener("click_b10", function () {
[... 235 lines cut ...]

              case 8:
                _context6.next = 10;
                return this.setText("t2", total + "mm");

              case 10:
              case "end":
                return _context6.stop();
            }
          }
        }, _callee6, this);
      }));

      function update(_x4) {
        return _ref6.apply(this, arguments);
      }

      return update;
    }()
  }]);

  return ZAxis;
}(_abstract3.default);

exports.default = ZAxis;

mylove29985, 2018-04-25:
Hi I tried to reinstall 2.4 tft file version 0.2. In this case Zaxis page work like a charm. So i think problem come from 2.4 tft file version 0.3 thanks

SteBuTOS, 2018-04-25:
Can you tell me what file-size the 2.4.tft of your 0.3 version has?

SteBuTOS, 2018-04-25:
...oh I now see the reason: somehow the nextion IDE reassigned the object-id's during downscaling. It seems only be the case for the zAxis screen, with effect for the 2.4, 2.4 enhanced, 2.8 and 2.8 enhanced tft's. Thats a bad situation, as the object ids are not editable (assigned by the IDE), but must match through all versions of HMI files to correspond with the javascript. Have to figure out a way to get that fixed.

SteBuTOS, 2018-04-25:
[2.4.tft.zip](https://github.com/Tibus/TouchScreen-For-NanoDLP/files/1948269/2.4.tft.zip)
can you please try this one

SteBuTOS, 2018-04-25:
(...of course you need to unzip it first).
Give a short feedback, if we have fixed it. In that case I need to make a pull request to provide the fix to Julien.

SteBuTOS, 2018-04-26:
Thanks for your feedback. I already made a pull request to Julien

Tibus, 2018-04-26:
Sorry was a bit occupied on other things. I just accept your pull request and will really try to make a Release today!

SteBuTOS, 2018-04-26:
Laser on fdm printer?

Tibus, 2018-04-26:
yeap ;D https://www.youtube.com/watch?v=QYVMZF7p-S4
And modified the LaserWeb4 repository, create a pull request to make this awesome software compatible

mylove29985, 2018-04-26:
Dear TOS I am waiting for next version with can choise Plate from usb. It made an awesome standalone machine without network connection. Happy to test next version on my own machine :D

SteBuTOS, 2018-04-26:
@mylove29985 as you downloaded a pre-release from my fork, you should already have that functionality available within your current setup. Just check for the plus-icon in the top right corner of the Plates-screen.

mylove29985, 2018-04-26:
@SteBuTOS I know but it not working any button and stuck on that page, cant back to home page like the last bug of Zaxis page so I think it just a demo version! So this the same problem like Zaxis page ?
https://upanhtocdo.com/image/Sqs7j

mylove29985, 2018-04-26:
Oh I know how to solve that just insert USB then all the Problem gone :D Wonderful!!!

SteBuTOS, 2018-04-26:
Did I get you right? The home button in the „add Plate“ screen does not work?

mylove29985, 2018-04-26:
Yes it work! I checked all and everything is working well 😀

polyfield, 2020-12-11:
hello,i am a chemist in 3d uv curing materials, i want a diy interface so i follow your guys tips to install this script,but after connecting its bottoms not working ,is there any wrong with my terminal inputs?   somebody can help or give me some information? 

pi@raspberrypi:~/nextion $ sudo npm i ip --unsafe-perm
screen@1.0.0 /home/pi/nextion
└── ip@1.1.5

npm WARN screen@1.0.0 No repository field.
pi@raspberrypi:~/nextion $ sudo npm i drivelist --unsafe-perm

> drivelist@6.4.6 install /home/pi/nextion/node_modules/drivelist
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=6.4.0 runtime=n                                                                                                                                                             ode arch=arm platform=linux)
make: Entering directory '/home/pi/nextion/node_modules/drivelist/build'
  CXX(target) Release/obj.target/drivelist/src/drivelist.o
In file included from /root/.node-gyp/6.4.0/include/node/node.h:42,
                 from ../../nan/nan.h:56,
                 from ../src/drivelist.cpp:17:
/root/.node-gyp/6.4.0/include/node/v8.h: In static member function ‘static void                                                                                                                                                              v8::V8::RemoveGCPrologueCallback(v8::GCCallback)’:
/root/.node-gyp/6.4.0/include/node/v8.h:8504:57: warning: cast between incompati                                                                                                                                                             ble function types from ‘v8::GCCallback’ {aka ‘void (*)(v8::GCType, v8::GCCallba                                                                                                                                                             ckFlags)’} to ‘v8::Isolate::GCCallback’ {aka ‘void (*)(v8::Isolate*, v8::GCType,                                                                                                                                                              v8::GCCallbackFlags)’} [-Wcast-function-type]
       reinterpret_cast<v8::Isolate::GCCallback>(callback));
                                                         ^
/root/.node-gyp/6.4.0/include/node/v8.h: In static member function ‘static void                                                                                                                                                              v8::V8::RemoveGCEpilogueCallback(v8::GCCallback)’:
/root/.node-gyp/6.4.0/include/node/v8.h:8511:57: warning: cast between incompati                                                                                                                                                             ble function types from ‘v8::GCCallback’ {aka ‘void (*)(v8::GCType, v8::GCCallba                                                                                                                                                             ckFlags)’} to ‘v8::Isolate::GCCallback’ {aka ‘void (*)(v8::Isolate*, v8::GCType,                                                                                                                                                              v8::GCCallbackFlags)’} [-Wcast-function-type]
       reinterpret_cast<v8::Isolate::GCCallback>(callback));
                                                         ^
In file included from ../../nan/nan.h:56,
                 from ../src/drivelist.cpp:17:
../src/drivelist.cpp: At global scope:
/root/.node-gyp/6.4.0/include/node/node.h:447:43: warning: cast between incompat                                                                                                                                                             ible function types from ‘void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’ {aka                                                                                                                                                              ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*                                                                                                                                                             )(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
       (node::addon_register_func) (regfunc),                          \
                                           ^
/root/.node-gyp/6.4.0/include/node/node.h:478:3: note: in expansion of macro ‘NO                                                                                                                                                             DE_MODULE_X’
   NODE_MODULE_X(modname, regfunc, NULL, 0)
   ^~~~~~~~~~~~~
../src/drivelist.cpp:66:1: note: in expansion of macro ‘NODE_MODULE’
 NODE_MODULE(DriveList, InitAll);
 ^~~~~~~~~~~
In file included from /root/.node-gyp/6.4.0/include/node/node.h:42,
                 from ../../nan/nan.h:56,
                 from ../src/drivelist.cpp:17:
/root/.node-gyp/6.4.0/include/node/v8.h: In instantiation of ‘void v8::Persisten                                                                                                                                                             tBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallba                                                                                                        [... cut ...]
    │   ├─┬ buffer-alloc@1.2.0
    │   │ ├── buffer-alloc-unsafe@1.1.0
    │   │ └── buffer-fill@1.0.0
    │   ├── fs-constants@1.0.0
    │   ├── to-buffer@1.1.1
    │   └── xtend@4.0.2
    ├─┬ tunnel-agent@0.6.0
    │ └── safe-buffer@5.1.2
    └── which-pm-runs@1.0.0

npm WARN screen@1.0.0 No repository field.
pi@raspberrypi:~/nextion $ sudo npm i child_process --unsafe-perm
screen@1.0.0 /home/pi/nextion
└── child_process@1.0.2

npm WARN screen@1.0.0 No repository field.
pi@raspberrypi:~/nextion $ sudo shutdown -r now