Tibus/TouchScreen-For-NanoDLP #15: Connecting to printer error [Rpi 2 B+ and nextion 3.5]

rdastartupguy, 2018-05-23:
Hi all, 
Trying to get this working with nanodlp but I have two concerns. 

1. I get the "connecting to printer" on my Rpi 2 B+ with nextion 3.5 screen setup 
2. I'm confused with the GPIO pin out diagram provided in the hardware readme section. On Raspberry.org it says GPIO 14 & 15 as Tx, Rx but in Tibus's reference diagram its written 15 & 16. Am I reading it wrong? 

Anyone who's gotten past the "Connecting to printer.." error? I tried everything in rpi3 issue but it doesn't work.

rdastartupguy, 2018-05-24:
Tried everything given in the forums + this one as well https://github.com/cuboid3d/nanodlp 
Still stuck at "Connection To Printer..." 

H/w -- Rpi2 B+ and Nextion 3.5

jamesarm97, 2018-05-24:
I had a long time ago. The first time was very tricky and I ended up using a USB to TTL serial adapter instead of the Rpi GPIO RX/TX and it worked. The second worked without it following the steps. You need to make sure when you run the pi server side that it is not reporting any errors. I did this by running the Nextion script on the console and watched it.

jamesarm97, 2018-05-30:
Correct about the usb to serial ttl. That is how I use mine, no messsing with the gpio. Just have to note and select the correct serial device in the config file (though the GUI or command line editing).

stewie0056, 2018-06-11:
Hello,
I had a Pi 2 B and Nextion 3.5 Basic with the not connecting issue too. The fix for me was to disable the shell over serial option.

 Type "sudo raspi-config"
 Select "Interfacing Options" 
 Select "Serial"
 Select "No"
 Select "Yes"
 Select "Ok"
 Select "Yes" to reboot

Hope this helps.

rdastartupguy, 2018-06-14:
Guys sorry for the delay in updating the status, yes I got it to work over rpi2. I had installed wrong script and then installed the correct script which may have caused conflict. I made a clean wipe, installed nanoDLP image and then proceeded with Tibus installation instruction and it worked. Thank you Stewie and James, without your inputs I wouldn't have made it. Much appreciated!

closed 2018-06-22 by rdastartupguy