Tibus/TouchScreen-For-NanoDLP #17: Please help me

arvin2009, 2018-07-24:
Hello
I did several steps.
The menu opens
But the touch key does not answer
The running time of the last stages
The following messages are displayed

unlink: cannot unlink '/usr/bin/node': No such file or directory
unlink: cannot unlink '/usr/sbin/node': No such file or directory
unlink: cannot unlink '/sbin/node': No such file or directory
unlink: cannot unlink '/usr/local/bin/node': No such file or directory
unlink: cannot unlink '/usr/bin/npm': No such file or directory
unlink: cannot unlink '/usr/sbin/npm': No such file or directory
unlink: cannot unlink '/sbin/npm': No such file or directory
unlink: cannot unlink '/usr/local/bin/npm': No such file or directory

Please list the new steps again

arvin2009, 2018-07-24:
My problem solved with this method

ok, had some time on my hand on a linux machine. the following script shourd work:sudo wget https://github.com/SteBuTOS/TouchScreen-For-NanoDLP/archive/v0.3.tar.gz --no-check-certificate -O - \| sudo tar -C /home/pi/nextion -xzsudo cp /home/pi/nextion/TouchScreen-For-NanoDLP-0.3/. /home/pi/nextion/. -rvsudo rm -r /home/pi/nextion/TouchScreen-For-NanoDLP-0.3sudo npm i ipsudo npm i drivelistsudo npm i child_processsudo shutdown -r now
--

WeiMingYu, 2018-11-15:
Pi3B install nodejs try again

wget https://nodejs.org/dist/v10.0.0/node-v10.0.0-linux-armv7l.tar.xz
xz -d node-v10.0.0-linux-armv7l.tar.xz
tar -xavf node-v10.0.0-linux-armv7l.tar
sudo ln -s /usr/local/node/bin/node /usr/bin/node
sudo ln -s /usr/local/node/bin/npm /usr/bin/npm
Check if it is installed
6. node --version
7. npm --version

run

sudo npm i ip sudo npm i drivelist
sudo npm i child_process
sudo shutdown -r now
Disable Bluetooth
1.sudo systemctl disable hciuart
2.sudo reboot

I am using NEXCION 7"

Some features are currently not working.

My solution is to modify the content of the program.

CaesarMaria, 2018-11-29:
Hi there, i get the same messages using a Pi2B any idea how i get nodejs up and running on that model?

arvin2009, 2018-11-30:
ok, had some time on my hand on a linux machine. the following script shourd work:

sudo wget https://github.com/SteBuTOS/TouchScreen-For-NanoDLP/archive/v0.3.tar.gz --no-check-certificate -O - | sudo tar -C /home/pi/nextion -xz

sudo cp /home/pi/nextion/TouchScreen-For-NanoDLP-0.3/. /home/pi/nextion/. -rv

sudo rm -r /home/pi/nextion/TouchScreen-For-NanoDLP-0.3

sudo npm i ip

sudo npm i drivelist

sudo npm i child_process

sudo shutdown -r now

CaesarMaria, 2018-12-01:
Thank you for your answer. I tried using your comands and got a lot of errors. I´ve looked these up and installed node.js and npm following this link:
https://www.instructables.com/id/Install-Nodejs-and-Npm-on-Raspberry-Pi/
if i check the versions of both, i get a correct respond. But if i try to follow your suggestions i get the following errors:

pi@raspberrypi:~ $ sudo npm i drivelist
                                                                                                        k for /home/pi
> drivelist@6.4.3 install /home/pi/node_modules/drivelist
> prebuild-install || node-gyp rebuild

prebuild-install WARN install EACCES: permission denied, access '/root/.npm'
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.4.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/drivelist/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/node_modules/drivelist/.node-gyp'
gyp ERR! stack     at Error (native)
gyp ERR! System Linux 4.14.79-v7+
gyp ERR! command "/opt/nodejs/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/drivelist
gyp ERR! node -v v6.4.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! drivelist@6.4.3 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the drivelist@6.4.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-12-01T09_29_06_549Z-debug.log

CaesarMaria, 2018-12-01:
I´m sorry i just looked into my answer and realised i didn´t post the answer that i get on your fist command. The answer is the following:

pi@raspberrypi:~ $ sudo npm i ip
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'                      home/pi/node_m
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'                         kage.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ ip@1.1.5
updated 1 package and audited 5388 packages in 65.581s
found 0 vulnerabilities