Tibus/TouchScreen-For-NanoDLP #18: Home screen hangs.

smagtech, 2018-09-13:
Hi I am Sudhir from India. I am using Raspberry Pi model 2, 1GB and nexon 3.2 enhance model. After displaying "connecting to printer...... "goes to home screen and hangs there. touch buttons are responsive but nothing happens. what I am doing wrong please help me.
I attached i2c LCD 16 by 2 it is working. 
[image]

mutsched, 2018-09-17:
Hello to you. If have same problem with 3.2 nextion rpi3b+. I read that the problem is at the defination from the buttons. But i also found no solution...

smagtech, 2018-09-20:
Thank you for replay. problem solved.  after this steps.....

For Raspberry pi2,1gb model and nextion3.2 enhanced.
/////////////////////////////////////////////////////////////////

step 1= upgrade os of Raspberry pi2.
------------------------------------------------------
sudo apt-get update 
sudo apt-get upgrade
.then>>
sudo reboot   
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

step 2= install touchscreen-for-nanodlp package.
------------------------------------------------------
wget -O - https://raw.githubusercontent.com/sdesalas/node-pi-zero/master/install-node-v6.4.0.sh | bash
.then>>
mkdir /home/pi/nextion;(wget https://github.com/Tibus/TouchScreen-For-NanoDLP/releases/download/V0.3/pi2_3.tar.gz --no-check-certificate -O - | tar -C /home/pi/nextion -xz);cd /home/pi/nextion; sudo ./setup.sh
.then>> ( important step )
sudo cp * / -rvf
.then>>
sudo reboot
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

step 3= change pi cmdline.text file
------------------------------------------------------
switch off Raspberry pi2. remove micro sd card. 
insert micro sd card into pc. 
find file ( cmdline.TXT ) open in Notepade++. 
and replace this.....
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=b94f2c65-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
with this.....
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
save and exit.

//////////////////////////////////////////////////////////////////