Tibus/TouchScreen-For-NanoDLP #20: PLS HELP I AM DESPAIR... RPI3B+ / NEXTION 3.2

mutsched, 2018-10-27:
Hello to all... I have problems to get the nextion running... I spend days of trys but i cant get it to worlk..
In the internet there are some manuals but the most dont work or i dont check it how to do.
So i decidet to maken an pdf MANUAL for beginners to install Nanodlp and Nextion on an brand new RPI3B+
But at moment my display shows only Connecting to printer...

What i did until now :

Download the new Nano DLP RPI Image from 	
https://www.nanodlp.com/download/

If you use Windows download ETCHER from
https://www.chip.de/downloads/Etcher-Image-to-USB-SD-Karte_96464610.html
and install it.

Unzip nanodlp and start ETCHER
Select Image you want to flash on your SD Card 
Select nanodlp and flash it

Download Putty from
https://www.putty.org/

Connect your RPI 3B+ with an ethernet cable to your router
After flashing sd put SD Card into the RPI 3B+ and start RPI

Go to your router interface ( in my case 10.0.0.138 ) and login
Watch in your Router Interface and look for RPI 3B+ ip-adress

Open Putty on PC and type in ip-adress from your RPI 3B+  and click open
Then you come to the putty interface box and you can login to your RPI 3B+

Login as: pi
pi@10.0.0.5´s password: raspberry
 
Now youre on the RPI 3B+
Type sudo raspi-config and make some settings in the “Raspberry PI Software configuration tool” 
The Changes debends on your system or on the country you live

 Got to 4. Localisation Options and change
	Change Locale to	“de_DE.UTF-8” (ONLY IN MY CASE)
	Change Timezone to	“EUROPE” and then “BERLIN”
	Change Keyboard layout

If all settings changed 	got to finisch and click enter

Now you see the black putty interface again. Now update the RPI3 B+ and type
	sudo apt-get update
	sudo apt-get upgrade
	sudo reboot

Congrats so your RPI3 B+ is in ground configuration !!!

Now Type the Ip-Adress from your RPI 3B+ into your webbrowser

Now you can log in to your NanoDLP interface on RPI 3B+
 
Here you can chose different types of preconfigured Printers. In my case I use the “mUVe3D” Version. Click on it and RPI 3B+ restarts

After Restart look if there is an “New Version Available” in the right upper corner.
If yes click on it and “Upgrade to the latest stable version”

Now the RPI 3B+ update and reboot again

GET NEXTION TOUCH DISPLAY WORK ?!?!?! ( Here I need Help !!!  )

Go to https://github.com/Tibus/TouchScreen-For-NanoDLP/tree/master/nextion/tft
And download the tft file for your display. (in my case the Nextion 3,2)

Put the tft file into an emty MICRO SD Card 
Put the card into the cardslot on your Nextion
Power on the nextion and it will update automaticly
If ready the display shows 100% success

Now get back on Putty and log in to your RPI 3B+ and check your serial0/serial1 connection

Type followed lines
cd /dev
ls- l

Now you have to scroll up and see followed picture
Serail1 -> ttyAMA0

Type
Sudo raspi-config
Go to  5 Interface Options
 
First question: 	Would you like a login shell to be accessible over serial  NO
Second Question: 	Would you like the serial port hardware to be enabled  YES

The prompt will read :	The serial login shell is disable 
				The serial interface is enabled

Select ok and allow the RPI3B+ to reboot 

Type
sudo nano /boot/config.txt
add fllowed line to the config.txt
	dtoverlay=pi3-disable-bt
exit and save changes

sudo reboot 

After reboot you log in over putty again and type
cd /dev
ls -l

Now it should show follow lines under serial 

Serial0 -> ttyAMA0
Serial1 -> ttyS0

Download Nextion Screen Software for Simple installation

Type
sudo mkdir /home/pi/nextion;(wget https://github.com/Tibus/TouchScreen-ForNanoDLP/releases/download/V0.2/pi2_3.tar.gz --no-check-certificate -O - | sudo tar -C /home/pi/nextion - xz);cd /home/pi/nextion; sudo ./setup.sh

NOW MY config.txt SHOWS

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
disable_camera_led=1
hdmi_pixel_encoding=2
start_x=1
gpu_mem=128
enable_uart=1
dtoverlay=pi3-disable-bt

MYcmdline.txt SHOWS

dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=b94f2c65-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

BUT MY DISPLAY ALWAYS SHOWS CONNECTING TO PRINTER….

What is wrong  ?? What changes I must make to get it work ???
THanks Daniel

WeiMingYu, 2018-11-15:
Pi3B install nodejs  try again

1. sudo npm i ip sudo npm i drivelist
2. sudo npm i child_process
3. sudo shutdown -r now

Disable Bluetooth

1.sudo systemctl disable hciuart
2.sudo reboot

mutsched, 2018-12-01:
Hello WeiMingYu sorry for late reply. First i have to say thank you for your answer and thank you that you take time to read my issue... So thanks for manual i will try tomorrow !!
Before your manual hav i change the config.txt with overlay ??

Or is at you manual a skipt at the nodejs what make the changes at config and cmdline ??

I ask because i want to make a clear manual to activate the nextion on RPI3B+ !! A manual for dummys because i am also not good in coding and i takes so much time to get the situation run for me !! I spend hours of hours on google and on youtube but i dont find any clear manual. I wan a to make a manual with pictures and comandolines for open source and i think much peaople are at the same Situation like me.....

so First have to change config.txt
Then change cmdline.txt
then upgrade and update rpi3b+
then install nextion programm from :
mkdir /home/pi/nextion;(wget https://github.com/Tibus/TouchScreen-For-NanoDLP/releases/download/V0.3/pi2_3.tar.gz --no-check-certificate -O - | tar -C /home/pi/nextion -xz);cd /home/pi/nextion; sudo ./setup.sh
then do the steps you write above.

Have i forget something ?? Or maybe anyone can give or show me a manual for dummys :)

adam440, 2019-02-03:
Hello,
any progress on this?

mutsched, 2019-02-04:
Hello to you :)Yes in the meantime i got to fix the problem. Now i am testing my diy printer :))
Do you also know something about printing wih LCD resin Printer ??
I have some problems with start up code from nanodlp and so on..
best regards

adam440, 2019-02-05:
I'm sorry, I can't help you yet. The hardware of my printer is almost done and now I'm trying to get the firmware working. I've got the Tibus software working in summer, but somehow, when I started the Raspberry Pi again, few weeks ago, it didn't work.

I tried couple of different strategies to get it to work again, but with no success. Could you please send me the disk binary copy (.img file)? I would be very glad, and if I somehow make my printer functional, I'll gladly help you with your printer.

Also, what exactly has you done to get the Tibus software working?

Thanks 

Adam Schuppler 
Best regards

Ackta, 2019-02-20:
Hi i want to install tibus on my phrozen
for customizing the nextion interface.

vl80k, 2021-01-19:
Hello! 
It there some progress with solution? 
Thank You!