Tibus/TouchScreen-For-NanoDLP #24: 3.2_Enhanced Does Not Match

ezryder645, 2019-02-11:
Put the 3.2_enhanced.tft on the SD card and when booted got the error message:"
Model does not match
Device Model:
NX4024K032_011R
"
Any idea what might be happening here?

SteBuTOS, 2019-02-11:
Try 3.2.tft. Your screen might not be the enhanced version.

ezryder645, 2019-02-20:
Thanks! I tried both and didn't work, but I rebuilt the ui in Nextion editor from the hmi file and it worked. Now it boots the main page but the buttons don't seem to work

HacknaTon, 2019-04-04:
sudo wget https://github.com/SteBuTOS/TouchScreen-For-NanoDLP/archive/v0.3.tar.gz --no-check-certificate -O - | sudo tar -C /home/pi/nextion -xz

sudo cp /home/pi/nextion/TouchScreen-For-NanoDLP-0.3/. /home/pi/nextion/. -rv

sudo rm -r /home/pi/nextion/TouchScreen-For-NanoDLP-0.3

sudo npm i ip

sudo npm i drivelist

sudo npm i child_process

sudo shutdown -r now

do that ;)